bgiot/Prusa-Firmware-BMG #1: can't print - changing filment urge

kobrok, 2019-03-24:
Hello,
my printer got through calibration successfully. Now every time I start printing process, it is stopping by pushing on to change filament after 7x7 probing. MK3S sensor + Bondtech + 0.9 LDO. I use BGIOT-MK3S-3.7.0rc1-BMG16-firmware.hex

bgiot, 2019-03-25:
What are you trying to print? Not an issue in the gcode ? Did you succeed to do the first layer calibration ?

kobrok, 2019-03-25:
It's not the gcode issue - as the same file is working with other firmware edition.

bgiot, 2019-03-25:
Can you try this version ? you need to unzip it.
[BGIOT-MK3S-3.7.0rc1-BMG16-firmware.hex.zip](https://github.com/bgiot/Prusa-Firmware-BMG/files/3003249/BGIOT-MK3S-3.7.0rc1-BMG16-firmware.hex.zip)

bgiot, 2019-03-27:
@kobrok Any news ? did you have time to apply this new updated firmware? Is calibration still ok? Can you print?

kobrok, 2019-03-30:
Hey, that new version seems to work. Although I had to introduce M92 E830 in the gcode to set esteps for 0.9. Thank you.

bgiot, 2019-03-30:
I changed the readme file about applying the eeprom update. In fact flashing the firmware doesn’t update the eeprom. 
So you confirm that calibration works and printing also?
Thanks for your feedback.

kobrok, 2019-03-30:
It's printing for one hour now - working fine so far.  I'll inform you if any issues occur. Thank's!

closed 2019-03-30 by bgiot