bgiot/Prusa-Firmware-BMG #3: Z height calibration fails

ogland, 2019-04-02:
Change #define Z_MAX_POS 210 to #define Z_MAX_POS 205 in order for z calibration to work. This change is for both mk2.5s and mk3s using bondtech extruder upgrade. If 205 fails too, lower it to 203. 205 works for mk2.5s on bear, and bear should add 1-2 mm z height.

This is because the new mk3s (and mk2.5s?) extruder has a higher pinda position than the bondtech extruder and mk3 extruder etc, so the new expected z height is a little bit different.

bgiot, 2019-04-02:
Another solution has been applied by changing calibration code to use MK3 compensation even with MK3S. This solution was proposed by G. Saunier.  Did you use the right firmware version? I don't support MK25S right now. Is it for MK25S you need this change ?

ogland, 2019-04-02:
I don't really understand what you are saying. In the readme you describe changes for mk2.5. Surely this firmware must support mk2.5s as 3.7.0 supports it.

bgiot, 2019-04-02:
No, MK2.5S is a different configuration/build of the firmware.
Ok, so your needs are related to MK2.5S version.
MK3S version is working fine I think. I'll create a new version for MK2.5S with modified Z_MAX_POS to 203.
I'll keep you posted.

ogland, 2019-04-02:
MK2.5s is just a different variants file, but it would probably be nice with a version with that too :)

bgiot, 2019-04-02:
I didn't create a BMG MK25S variant in fact ;-) That's why. I'll create it.

bgiot, 2019-04-02:
Do you need it for 1.8° or 0.9° stepper

ogland, 2019-04-02:
I'm using a 0.9 stepper personally. My esteps are tuned at 810 so no need for a special one for me, except for the F2400 max feedrate for E. Users with 1.8 motor can of course do double that.

bgiot, 2019-04-02:
[BGIOT-MK25S.13a-3.7.0-BMG16-firmware.zip](https://github.com/bgiot/Prusa-Firmware-BMG/files/3034107/BGIOT-MK25S.13a-3.7.0-BMG16-firmware.zip)

Can you try this ?

ogland, 2019-04-02:
Sorry no. I need to see changes and compare it to what I'm running before I dare update firmware again. I just got it working a couple of days ago.

bgiot, 2019-04-02:
https://github.com/bgiot/Prusa-Firmware/blob/BMGCustom/Firmware/variants/MK25S-RAMBo13a-BMG-16.h

bgiot, 2019-04-05:
@ogland  maybe you can also share your changes…

bgiot, 2019-05-09:
No feedback provided. Closing this issue as other people succeeded with calibration.

closed 2019-05-09 by bgiot