bgiot/Prusa-Firmware-BMG #4: MK25s with inverted extruder direction?

zawadzkip, 2019-04-03:
Any chance you can make a variant of the MK2.5S build with
```c
#define INVERT_E0_DIR 0    // for direct drive extruder v9 set to 1, for geared extruder set to 0
#define INVERT_E1_DIR 0    // for direct drive extruder v9 set to 1, for geared extruder set to 0
#define INVERT_E2_DIR 0    // for direct drive extruder v9 set to 1, for geared extruder set to 0
```

having some issues compiling :/

zawadzkip, 2019-04-03:
Did manage to get it to compile, if you'd like I can submit a PR, but it's minor changes.

bgiot, 2019-04-03:
Did you create a variant? For which printer is it?

bgiot, 2019-04-03:
If you use the cable provided by Bondtech, you don't need to change the E motor direction…

zawadzkip, 2019-04-03:
@bgiot I will double check on the cable. I may have missed that part during assembly but I believe I am using that cable.... But yes I believe the BMG + MK2.5S needs the inversion.

https://www.bondtech.se/en/customer-service/faq/ Note the "extruder turning in the wrong direction"

bgiot, 2019-04-04:
@zawadzkip You maybe need to swap some cables.  Many people confirmed that you just have to put cables in the right order to fix your issue.  This will avoid to have another firmware build to maintain.
Regards.

closed 2019-04-05 by zawadzkip