bgiot/Prusa-Firmware-BMG #6: Filament sensor on mk3s firmware does not work

Truetech000, 2019-06-01:
As stated, the sensor is not working on the custom firmware for mk3s 1.8

bgiot, 2019-06-01:
Which release did you try ?

Truetech000, 2019-06-02:
Mk3s 1.8 degree step

Truetech000, 2019-06-02:
Latest

bgiot, 2019-06-02:
Did you try the stock firmware for MK3S (make sure to set e-steps to 830) ? is IR sensor working?
Because I checked again the different configs for each firmware version build and all seem ok... maybe you have a hardware issue ?  As I don't have the S upgrade on my MK3, I cannot do any testing... sorry, I'll need your help to figure-out what's going on...

Truetech000, 2019-06-03:
It does work with stock, I am curious what all is changed with your fw
other than e steps and z distance?

bgiot, 2019-06-03:
I checked and only e-steps, z distance, and speeds for load/unload are changed....
When my firmware installed, does the screen displays 'Prusa BMG MK3S' ?

Truetech000, 2019-06-05:
I'm sorry, I found the issue. It was that the magnets in the add-on from
bondtech are too far apart in their design, I will be modifying it to make
them engage.

bgiot, 2019-06-05:
So, wait your confirmation that my custom firmware works fine to close the issue

Truetech000, 2019-06-05:
Yes, thanks again and sorry for the inconvenience

bgiot, 2019-06-05:
Issue closed because not an issue

closed 2019-06-05 by bgiot