bgiot/Prusa-Firmware-BMG #7: Stepper Motor Hot

bobbyromeo, 2019-06-28:
After installing the BondTech Prusa I3 MK2.5 MK3 Extruder Upgrade Kit on my MK3S. I notice the stepper motor (17hs10-0704s) gets very hot to the touch. 

I am using Prusa fimware BGIOT-MK3S-BMG-1.8-3.7.2-firmware.hex.

Bondtech supports suggests M907 E600 in starting gcode to lower the current but they're unsure whether it gets overwritten via the firmware. When trying it, I did not notice a difference.

Perhaps you can shed some light on this?

bgiot, 2019-06-29:
The firmware doesn't override any epprom or gcode settings... gcode settings are the final value.
The E-step motor current is set to 30 on a range of 0..63 (Prusa default config).  63 means the MAX current (don't know what it is). I'm running the same firmware right now, e-motor is warm but not too hot... Maybe ask question to the facebook prusa community page... I'll do few research on my side. If you have more info, please keep me posted!  
regards,

Bertrand.

bgiot, 2019-06-29:
Don't forget to M500 to save setting in eeprom

bgiot, 2019-06-29:
The 17hS10-0704S is 0.7A/phase. So 1.4A (2 phases?) => as 30/64 is 46% => 0.796A
The M907 command is setting absolute mA... You can try lower more...

bobbyromeo, 2019-06-29:
The M913 gcode to show current values is commented out in the source code: https://github.com/prusa3d/Prusa-Firmware/blob/c43bd9152050449f526e570f9fe72e7a4e3c5f44/Firmware/variants/1_75mm_MK3S-EINSy10a-E3Dv6full.h#L266-L270

https://github.com/prusa3d/Prusa-Firmware/blob/aff020a6/Firmware/Marlin_main.cpp#L6928-L6931

I was thinking to lower the scalar value in the code and then recompile but when I saw the 30 scalar value I wasn't sure how to convert to get 600 mAh based on that value. The code has a function to do so but not quite sure how to intepret it: https://github.com/prusa3d/Prusa-Firmware/blob/27db32cbd0d4f8ff7367e4609cc571a7567407fa/Firmware/tmc2130.cpp#L1001-L1021

So the current is saved in the eeprom when running the command M907 E600 ? I don't think I did a M500 after which may explain why I didn't see a difference. Right?

bgiot, 2019-06-29:
Sure you have to do M500 to save value in eeprom.

bobbyromeo, 2019-06-29:
Hmm. I'm not sure I understand. 

When I ran M907 E600 it seemed to raise the scalar value from 30 to 35

```
Send: M913
Recv: tmc2130_print_currents()
Recv: 	H	R
Recv: X	16	16
Recv: Y	20	20
Recv: Z	35	35
Recv: E	30	30
Recv: ok
[...]
Recv: MMU not responding - DISABLED
[...]
Send: M907 E600
Recv: ok
[...]
Send: M500
Recv: echo:Settings Stored
Recv: ok
Send: M913
Recv: tmc2130_print_currents()
Recv: 	H	R
Recv: X	16	16
Recv: Y	20	20
Recv: Z	35	35
Recv: E	36	36
Recv: ok

```

bgiot, 2019-06-30:
From the code you mentioned, there are two linear scale depending of the current value.  From 0 to 539 mA, the scale 0 to 63 is from 0 to 1157mA (1029 * 1.125).  For current from 540 to 1029mA, the 0..63 scale equals 0..1029mA.  So 600mA is equal to 36 indeed... 
Now the question is the value 30 is equal to ? 
Let's do the exercice with the second function in the code that is converting the value to mA:
30 => vsense = 1 => vfs = 0.18
((30+1)/32) * (0.18/(0.2+0.02) * 0.7071 = 0.560 => 560mA
And with 36?
36 => vsense = 0 => vfs = 0.32 => 36 must be devided by 2!
((18+1)/32) * (0.32/(0.2+0.02) * 0.7071 = 0,610 => 610mA
**Conclusion**: default setting (30) is 560mA, so lower than the 600 you set.

bobbyromeo, 2019-07-04:
Ok. Thanks for your help.

closed 2019-07-04 by bobbyromeo