bgiot/Prusa-Firmware-BMG #8: Z height after calibration

bobbyromeo, 2019-07-06:
Using your firmware Z_MAX_POS is set to 204, which satisfies calibration. I need slighly higher Z height. Would it be Okay if I simply reset Z_MAX_POS 210 and reflash my MK3S? Would this cause any issues you think?

I"m using vekoj's X-carriage https://www.thingiverse.com/thing:3560116/comments with the MK3S sensor addon.

bgiot, 2019-07-06:
Should be fine unless you re-do calibration... just try and let me know.

bobbyromeo, 2019-07-07:
Seems fine. Will need to remember to reflash when I redo a calibration. Thanks.

bobbyromeo, 2019-07-07:
Odd thing, after printing to a hieght of 205, the head parked as per usual but for some reason it triggered a auto home of both X and Y. Then I noticed that octoprint lost communication with it. Not sure if it had to do with the height of 205 or not.

bgiot, 2019-07-07:
I don't really know... should not be linked... sorry, I'll not be able to help you. Maybe ask the community on facebook...

bgiot, 2019-07-08:
If you want, I made new build of the firmware with the height set to 210 and another fix at the calibration level to get around calibration issue.  Not yet published, but tell me which version you need (MK25/MK25S/MK3/MK3S)... I'll drop it in the discussion.

bobbyromeo, 2019-07-08:
MK3S. Thanks.
Can I just clone from https://github.com/bgiot/Prusa-Firmware/tree/BMGCustom and build myself?

bgiot, 2019-07-08:
Sure you can clone. But I already update the hex file in the release.  Also I did update e-steps configuration according to community recommendation.  So, you'll have to update eeprom settings (see the readme on this repo)

bobbyromeo, 2019-07-08:
Ok thanks. I'll clone from your BMGCustom branch and copy over the MK3S-BMG-1.8.h. Why the change to 415 from 830 for the Bondtech 1.8 motor?

bgiot, 2019-07-08:
Seems that the einsy board cannot handle over 500 without loosing steps few times.  I found also that you need to run the command in the right order. Seems that M350 divides by 2 the value of M92. So run M350 before M92 as explained in the readme.

bobbyromeo, 2019-07-08:
I'm getting an odd clicking sound with the firmware. How would I go back to the previous firmware with 830? 

M350 E32
M92 E830
M500
?

closed 2019-07-15 by bobbyromeo