bgiot/Prusa-Firmware-BMG #9: Extruder Clicking/Rattling on Retractions

bobbyromeo, 2019-07-15:
I'm noticing an odd clicking / rattling sound coming from the extruder. It appears to happen on
retractions mostly. Is this sound normal? Is it due to the PLA rubbing on a
part of the extruder? Or perhaps the gears meshing together?

I've uploaded a short video demonstrated this sound: https://streamable.com/30de7

According to Bondtech support, "The linear advance used with original Prusa firmware will create this sound due to the gearing ratio, try by disabling the Linear Advance and the sound should go away. There is a new version that implements the latest version of the Linear advance that will work much better."

bgiot, 2019-07-15:
This sound is normal. Indeed linked to retraction. You can eventually reduce the retraction distance... LA 1.5 should solve this indeed...

bobbyromeo, 2019-07-15:
Thanks again.
Where is this changed?
The line in PrusaSlicer?

M900 K{if printer_notes=~/.*PRINTER_HAS_BOWDEN.*/}200{elsif nozzle_diameter[0]==0.6}15{else}30{endif} ; Filament gcode

bgiot, 2019-07-16:
Indeed this setting must be change to cope with LA 1.5.  Which LA1.5 firmware did you take ?

bobbyromeo, 2019-07-16:
I'm not sure what you mean. I compiled the firmware based on your https://github.com/bgiot/Prusa-Firmware/tree/BMGCustom branch.
How do you change LA to 1.5 in the code?

bgiot, 2019-07-16:
It's not just config... search the facebook community group to find LA15 custom firmware. Mine doesn't have the custom part for LA15

bobbyromeo, 2019-07-16:
I'm looking...having trouble finding it.

bgiot, 2019-07-17:
I think so, if you have an MK3**S**

closed 2019-07-18 by bobbyromeo